pimoroni/pimoroni-pico #1015: update tutorial for the Visual studio code extention

bendelathouwer, 2024-10-12:
Would it be posible to make a tutorial installin the pimoroni c libaries using the new visual studio code extention?

helgibbons, 2024-10-15:
Not sure what extension you're referring to here, please can you post a link?

bendelathouwer, 2024-10-15:
https://www.raspberrypi.com/news/pico-vscode-extension/ this is the extention i'm talking about.

bendelathouwer, 2024-10-16:
been digging around a bit more and found that the makefiles the plugin generate are vastly diffrent that the ones you provide
[CMakeLists.txt](https://github.com/user-attachments/files/17400210/CMakeLists.txt)(generated by the plugin) vs yours (set(OUTPUT_NAME badger2040_drawing)
add_executable(${OUTPUT_NAME} badger2040_drawing.cpp)

target_link_libraries(${OUTPUT_NAME}
        badger2040
        hardware_spi
)

# enable usb output
pico_enable_stdio_usb(${OUTPUT_NAME} 1)

pico_add_extra_outputs(${OUTPUT_NAME})
)

bendelathouwer, 2024-10-16:
and sorry for the copy past but somehow github doesn't let me input the files from your repo. But also tbh i never worked with cmake so it might be that i'm missing something incredbly obvious