pimoroni/pimoroni-pico #1026: v1.24.0 - Migrating to PicoVector 2

Gadgetoid, 2024-10-31:
If you've managed to early-adopt PicoVector despite the lack of documentation and examples then you might notice *everything is broken* in v1.24.0. We've overhauled PicoVector hopefully for the better, and this necessitated some API changes. Sorry!

If you need help migrating over, then in lieu of updated documentation (it'll come, but the API isn't quite stable yet) please chime in here!

insanire-monachus, 2024-11-01:
I was just about to bring that up -- using several of the 1.23.x examples. Of course like you noted, broken..

[image]
(( fails to register.. etc. ))

berlincount, 2024-12-19:
Hello,

please port

https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/examples/tufty2040/polygons.py
https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/examples/tufty2040/vector_clock.py
https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/examples/tufty2040/vector_spectrometer.py

... just got the device - and wouldn't even know where to start without reading all of the docs :-D

cheers,

   count