pimoroni/pimoroni-pico #1028: SCD41 Breakout low-power mode not activated

marc-git, 2024-11-17:
Greeting, 
it seems that the SCD-41 should have a low-power version available according to [Sensirion documents](https://sensirion.com/media/documents/077BC86F/62BF01B9/CD_AN_SCD4x_Low_Power_Operation_D1.pdf), which would help a lot when monitoring CO2 from battery-run devices.

Any interest in developing that functionality in the python module?

Gadgetoid, 2024-11-18:
https://github.com/pimoroni/pimoroni-pico/pull/1029

Grab a build from here and try `start_low_power`, I haven't been able to test it but it's just calling one of Sensiron's API functions so it should work!

marc-git, 2024-11-22:
@Gadgetoid Thanks a lot! I can see exactly what needs doing but I am not experienced enough to compile an uf2 file yet i.e. my own badger os plus micropython. I did get the hello_world.c file compiled though. Perhaps it isn't so complicated but it will take me time to figure out without a walkthrough

marc-git, 2024-11-22:
ok well that was a can of worms. It seems the Badger UF2 files are not intended to be user-compilable. I've commented on an issue there to encourage that.

marc-git, 2024-11-22:
I got the micropython code to compile no problems though...

Gadgetoid, 2024-11-26:
Ooof, yes, distilling these changes down to Badger OS might be quite tricky since I haven't approached how much damage the Pico SDK 2.0.0 and 2.1.0 releases have done to that codebase yet 😬

I am slowly trying to move away from the GitHub actions workflows to build scripts regular folks (including myself) can follow. That's on the pile of nice-to-have, though, which is buried somewhere underneath all the AAAAAAAAAA.

Gadgetoid, 2025-01-16:
If you grab one of the Badger builds here it should include the new low power mode: https://github.com/pimoroni/badger2040/pull/91

Slowly working toward getting everything merged down, tested and released.

closed 2025-01-16 by Gadgetoid

marc-git, 2025-01-16:
mental, literally just had the commits open to see if it had been migrated! Thanks a lot.

marc-git, 2025-01-16:
@Gadgetoid is it your expectation that it will work with the breakout_scd41 python module? I still only see the following:

- start
- stop
- get_temperature_offset
- init
- measure
- ready
- set_ambient_pressure
- set_sensor_altitude
- set_temperature_offset

but they were all there before.

Gadgetoid, 2025-01-16:
There should be a new `start_low_power` but you'll need the pre-release build from the linked PR.

Will prod it and double check!

Gadgetoid, 2025-01-16:
Ah yup, I see it too (Or don't, as it were). Apparently I completely failed somewhere, bear with me!

marc-git, 2025-01-16:
thanks! works well now