pimoroni/pimoroni-pico #1031: More bitmap fonts

Slion, 2024-12-03:
Any chance we could get more bitmap fonts? 

So far I was happy using font8 on Display Pack and Display Pack 2 with a scale of 2. However on Display Pack 2.8 scale 2 is way too big and and scale 1 is way too small. So we basically don't have a bitmap font that's resonable for Display Pack 2.8.

Any chance we could get bitmap fonts of size 10, 12 and 14 maybe?

Gadgetoid, 2024-12-03:
We wont be packing any additional ones into the build, but I can probably convert some and whip up some documentation for how to use them. Though I don't know when I'll get a chance 😬

The winds of change are blowin' towards just converting regular fonts and using them with PicoVector, but that's still in development - https://github.com/pimoroni/pimoroni-pico/pull/1019

Slion, 2024-12-03:
Just wanted to put a reference to that other issue and comment here:
https://github.com/pimoroni/pimoroni-pico/issues/999#issuecomment-2513688219

peardox, 2024-12-15:
Yeah, that was a while ago... I got pretty far into highly alpha #999 as I also found problems with fonts on the 2.8 device

I sorta balked when I got into how basic bitmap fonts are in their data format. The encoding is extremely wasteful with every char using the max-char sizing.

I've been looking into fonts pretty deeply and the way they are rendered is much more efficient (and as with everything the documentation sucks). The data for a Glyph of any char only uses the actual pixels drawn, the rest is not coded as offsets are applied.

The larger of the grabs in #999 is for a 32pt Windows font - the 'pink bits' are the actual char data while the gray bits are just wasted. That font would take 13,860 bytes to store which sorta freaked me in Pico-terms. I did try storing the font out of RAM which did help and didn't hit performance too much. 

Guessing (never bothered counting real usage) about 50% of the data would be unused by my example (could be better/worse). 

For compatibility the data would have to be written as per existing methods I guess. 

While space could be saved via offsets and variable length recs this only helps larger point sizes, it would actually have the reverse effect on smaller fonts - catch-22

Think I might resurrect this as it is useful (would know the waste that way at least) - note app would not be available on Pi owing to the frameworks I'm using (everything else - just no ARM compiler for Linux on my preferred build language for this sorta thing)