pimoroni/pimoroni-pico #1035: I75W fails to activate wireless interface using MicroPython v1.24.0, picow v1.24.0-beta1

MikeB5027, 2024-12-07:
I have two recently-purchased Interstate 75 W boards, which both fail to activate their wireless network interface when using the latest Pimoroni picow Micropython release:
```
MicroPython v1.24.0, picow v1.24.0-beta1 on 2024-10-31; Raspberry Pi Pico W with RP2040
Type "help()" for more information.
>>> import network
>>> wlan = network.WLAN(network.STA_IF)
>>> wlan.active(True)
[CYW43] Failed to start CYW43
>>> print(wlan.active())
False
>>> 
```
Whereas the network interface was successfully activated with vanilla Micropython:
```
MicroPython v1.24.1 on 2024-11-29; Raspberry Pi Pico W with RP2040
Type "help()" for more information.
>>> import network
>>> wlan = network.WLAN(network.STA_IF)
>>> wlan.active(True)
>>> print(wlan.active())
True
>>> 
```
I raised this on the Pimoroni forum and tried all of Hel's suggestions, without success.

After nuking the I75W flash and reloading the latest Pimoroni picow MP image, it didn't produce the error message, but still failed to activate the wireless interface.

MikeB5027, 2024-12-09:
It seems that wlan.active() returns False even when the network interface is active.

It returns True once the network is connected, mimicking the behaviour of wlan.isconnected():
```
MicroPython v1.24.0, picow v1.24.0-beta1 on 2024-10-31; Raspberry Pi Pico W with RP2040
Type "help()" for more information.
>>> import network
>>> wlan = network.WLAN(network.STA_IF)
>>> wlan.active(True)
>>> print('wlan.active():', wlan.active())
wlan.active(): False
>>> from secrets import WIFI_SSID, WIFI_PASSWORD
>>> wlan.connect(WIFI_SSID, WIFI_PASSWORD)
>>> print('wlan.status():', wlan.status())
wlan.status(): 3
>>> print('wlan.active():', wlan.active())
wlan.active(): True
>>> print('wlan.isconnected():', wlan.isconnected())
wlan.isconnected(): True
>>> wlan.disconnect()
>>> print('wlan.active():', wlan.active())
wlan.active(): False
>>> print('wlan.isconnected():', wlan.isconnected())
wlan.isconnected(): False
>>> 
```

MikeB5027, 2025-01-15:
No, wlan.active() still reports False after activation but before connection:
```
MicroPython v1.24.0, picow v1.24.0-beta2 on 2024-12-11; Raspberry Pi Pico W with RP2040
Type "help()" for more information.
>>> import network
>>> wlan = network.WLAN(network.STA_IF)
>>> wlan.active(True)
>>> print('wlan.active():', wlan.active())
wlan.active(): False
>>> 
```
Thanks for looking :o)

djaychela, 2025-03-11:
Is there a fix for this?  I'm seeing the same having just got an I75W and a couple of boards today - I get `[CYW43] Failed to start CYW43` when I run the code from the demo repo (or other similar code to activate it).

helgibbons, 2025-03-11:
Do you have the most recent version of the Interstate 75 W MicroPython firmware installed? 

I75 now has its [own separate repo](https://github.com/pimoroni/interstate75) and MicroPython builds are downloadable from [the releases page](https://github.com/pimoroni/interstate75/releases).

helgibbons, 2025-03-11:
`wlan.active()` seems to be behaving itself for me, on the most recent Interstate 75 W builds:

[image]

[image]

djaychela, 2025-03-11:
Yes - got it this morning and used `i75w_rp2350-v0.0.5-micropython-with-filesystem.uf2` from the repo to get up and running (when I found this was playing with the `today.py` file....   And of course I go to try it now and it works!  I'll see if it's reliable as I'll be playing around with it, but thanks for getting back to me.