pimoroni/pimoroni-pico #1050: PicoGraphics - measure_text measures long

ahnlak, 2025-01-16:
When using `measure_text`, it includes any spacing applied to the final letter. This is a pretty minor problem until you start working at higher values for `scale` and `spacing`.

As a crude example, consider a spaced out string of 'Pimoroni': as things currently stand, the function will measure `P  i  m  o  r  o  n  i  ` (with the trailing space), while a more useful measure (for things like centering) would be `P  i  m  o  r  o  n  i` (sans trailing space).

I thought I'd raise an issue for a sanity check before I actually starting changing the code :-)

insanire-monachus, 2025-01-17:
Nope!!!.. you are not insane. I would agree.

ahnlak, 2025-01-18:
Fix in. I'm assuming there's a terribly clever way @Gadgetoid can blend that into the eleventy-million branches 🤣