pimoroni/pimoroni-pico #1054: Can't build the examples for latest pico-sdk

lerouxb, 2025-01-26:
[  4%] Building CXX object examples/breakout_encoder_wheel/clock/CMakeFiles/encoderwheel_clock.dir/clock.cpp.o
~/src/pimoroni-pico/examples/breakout_encoder_wheel/clock/clock.cpp:6:10: fatal error: hardware/rtc.h: No such file or directory
    6 | #include "hardware/rtc.h"
      |          ^~~~~~~~~~~~~~~~

lerouxb, 2025-01-26:
Got to the end by commenting out examples that wouldn't compile for one reason or another:

```
~/src/pimoroni-pico/build$ git diff
diff --git a/examples/CMakeLists.txt b/examples/CMakeLists.txt
index ae93c568..9e2607e8 100644
--- a/examples/CMakeLists.txt
+++ b/examples/CMakeLists.txt
@@ -1,6 +1,6 @@
 add_subdirectory(breakout_dotmatrix)
 add_subdirectory(breakout_encoder)
-add_subdirectory(breakout_encoder_wheel)
+#add_subdirectory(breakout_encoder_wheel)
 add_subdirectory(breakout_ioexpander)
 add_subdirectory(breakout_ltr559)
 add_subdirectory(breakout_colourlcd160x80)
@@ -47,8 +47,8 @@ add_subdirectory(pico_trackball_display)
 add_subdirectory(pico_audio)
 add_subdirectory(pico_wireless)

-add_subdirectory(inky_pack)
-add_subdirectory(inky_frame)
+#add_subdirectory(inky_pack)
+#add_subdirectory(inky_frame)

 add_subdirectory(automation2040w)
 add_subdirectory(plasma_stick)
@@ -60,7 +60,7 @@ add_subdirectory(servo2040)
 add_subdirectory(motor2040)
 add_subdirectory(inventor2040w)
 add_subdirectory(encoder)
-add_subdirectory(galactic_unicorn)
+#add_subdirectory(galactic_unicorn)
 add_subdirectory(gfx_pack)
-add_subdirectory(cosmic_unicorn)
-add_subdirectory(stellar_unicorn)
+#add_subdirectory(cosmic_unicorn)
+#add_subdirectory(stellar_unicorn)
```

Gadgetoid, 2025-02-21:
I've opened a PR to sniff out what breaks here: https://github.com/pimoroni/pimoroni-pico/pull/1058

Sorry for dropping the ball here. There have been quite a few changes, and I've been a little too buried in MicroPython.

closed 2025-05-29 by Gadgetoid