pimoroni/pimoroni-pico #1056: SDC41 Micropython missing control over auto-calibration and calibration commands

marc-git, 2025-02-21:
I'm trying to develop a device with the Badger to use the SDC41 as a battery-powered, standalone CO2 measurement system (not a commercial project). 

A well-documented problem with the SDC41 is the autocalibration feature which is turned on by default and more-or-less means that the device will take the lowest value over the last week as the new reference value. This makes it really unusable for a device that spends all of its time indoors. 

A number of functions are available in the given driver [information](https://sensirion.com/media/documents/077BC86F/62BF01B9/CD_AN_SCD4x_Low_Power_Operation_D1.pdf) and reference [manual](https://www.sensirion.com/resource/datasheet/scd40/scd41):

-     SCD4X_SET_AUTOMATIC_SELF_CALIBRATION_INITIAL_PERIOD_CMD_ID = 0x2445, 
-     SCD4X_SET_AUTOMATIC_SELF_CALIBRATION_STANDARD_PERIOD_CMD_ID = 0x244e, 
-     SCD4X_PERFORM_FORCED_RECALIBRATION_CMD_ID = 0x362f,

which are all present in the `scd4x_i2c` driver.  Would it be possible to include these in the micropython driver? As motivation: The usability threat is seriously critical over longer periods, and using the CO2 sensor to detect when a room needs to be aired makes the application quite important. Otherwise it is a 50€ toy that I cannot take seriously. 

I tried doing I2C directly via micropython and couldn't get it working.

Gadgetoid, 2025-02-21:
> Otherwise it is a 50€ toy that I cannot take seriously.

I appreciate the desire to light a proverbial fire under my tender behind, but this could have gone unsaid. It's not like a bright rainbow website full of pirate jokes screams "serious professional sensor equipment." 😆 

That said you make a compelling case. I'm grappling with Sensiron's cursed GitHub repository to bump our submodule before I add these features, but if it goes as smoothly as the last time (it's currently not 🤦* ) then it should be a case of wrapping a couple of their existing API functions.

(* Looks like they've wholesale stomped over their entire repository with the output of a "driver generator" and the result is about as unpleasant as you could imagine...)

Gadgetoid, 2025-02-21:
*backs away slowly from Sensiron GitHub*

Okay I've wrapped up the auto calibration features so they should be available to MicroPython. CI builds are here: https://github.com/pimoroni/pimoroni-pico/pull/1057

Forced recalibration probably isn't all that useful without a controlled environment and a good reference source - such is the eternal plight of uncalibrated sensors - but it's there if you want to get fancy.

marc-git, 2025-02-21:
> > Otherwise it is a 50€ toy that I cannot take seriously.
> 
> I appreciate the desire to light a proverbial fire under my tender behind, but this could have gone unsaid. It's not like a bright rainbow website full of pirate jokes screams "serious professional sensor equipment." 😆

Yes, you're right. I'm not one to demand resources in an open source context. My intention was more to give one the ammunition to justify working on it. Sorry for coming across entitled. 

> Forced recalibration probably isn't all that useful without a controlled environment and a good reference source - such is the eternal plight of uncalibrated sensors - but it's there if you want to get fancy.

Thanks a lot! 

To give an idea of how bad this is, my sensor is reading 700ppm for fresh air. Even a bad reference source (e.g. fresh air @ ~420ppm)  will be better. Sensirion engineers say an accurate reference calibration at 500ppm will have a +/- 40 ppm tolerance. 

I guess the Badger won't get the Micropython build for a while, right? I might have to bite the bullet and learn how to do these github workflows.

Gadgetoid, 2025-02-24:
> I guess the Badger won't get the Micropython build for a while, right? I might have to bite the bullet and learn how to do these github workflows.

Ah yes it takes *some time* to distill down. I'm sure I expedited some SCD4X changes there before, though, let me check.

Gadgetoid, 2025-02-24:
Okay assuming I haven't messed something up (I've probably messed something up) then these builds should contain the SCD4X changes: https://github.com/pimoroni/badger2040/pull/91

I think we're a little overdue a release for Badger!

Gadgetoid, 2025-02-24:
> Thanks for making my toy project possible

Any time. Just doin’ what I do!

(And also if you don’t complain about Badger 2040 exploding into flames I can probably merge those changes and release a new build 🤣)

marc-git, 2025-02-25:
will check asap. This week might be tricky

marc-git, 2025-02-27:
no flames on the 2040W or 2040. Calibration commands are there, and seem to work. Thanks.