pimoroni/pimoroni-pico #1066: Some CircuitPython Examples Please?

ChrisKahrs, 2025-03-18:
I love your products and I know you have a good relationship with AdaFruit.  I often am trying to use both at the same time and I struggle with their examples all being circuitpython and yours all being micropython.  Is there a way to create some of your examples with circuitpython?  Specifically I am struggling right now with the pico_lipo_shim as it defines a pin (gp24 on Pico) that I can't find what that is in CircuitPython.  Love your stuff so much.  Thanks a ton!

helgibbons, 2025-03-19:
I think you should be able to do:
```
import board
dir(board)
```
in CircuitPython to find out your board's pin names?

There's a small collection of CircuitPython examples over [here](https://github.com/pimoroni/pico-circuitpython-examples), but we're a small team with only so many hours in the day so need to make one ecosystem our primary focus, alas.

ChrisKahrs, 2025-03-19:
Totally understand.  Thank you for the pointers!