pimoroni/pimoroni-pico #1074: pimoroni_bus SPIBus not working in 1.25.0-beta1

jef41, 2025-04-24:
Using a Pico W board, with the firmware file ` pico-v1.25.0-beta1-pimoroni-micropython.uf2` when I try to create an SPIBus I always get the response `ValueError: invalid pin`. I am following the documentation at
https://github.com/pimoroni/pimoroni-pico/tree/v1.25.0-beta1/micropython/modules/picographics#spi--parallel

A minimal example; 

```python
from pimoroni_bus import SPIBus
spibus_L = SPIBus(cs=17, dc=16, sck=18, mosi=19)
spibus_R = SPIBus(cs=9, dc=8, sck=10, mosi=11)
```
The above works when using `picow-v1.24.0-beta2-pimoroni-micropython.uf2`, but reports an error on 1.25.0-beta1.

I get the hint from 'beta' in both those UF2 files and am possibly out of my depth here. I have spent some time trying alternate pin numbers or passing machine.Pins, but without success. I am going through commits between these releases, but have, so far (with limited C knowledge), failed to track down the cause. Hints to further diagnose (or resolve) gladly received.

Gadgetoid, 2025-04-24:
I must have broken it somehow in https://github.com/pimoroni/pimoroni-pico/commit/ba0bf0230d01f0e70bfca4beffa4485ea5c6e1d7

Will look into it ASAP!

jef41, 2025-04-24:
blimey, that was quick, thank you! 

I can confirm that if I roll back that change you reference and rebuild from the 1.25.0-beta1 branch, then the SPIBus example above does work again. No idea what/if anything else subsequently breaks, but I can run, for example, balls_demo.py on custom SPI pins.

Gadgetoid, 2025-04-25:
Apparently today is not my day, but it seems to be fixed now 🤦 

With any luck it'll make it to a release sooon!

jef41, 2025-04-25:
well it is evidently a good day for me, thank you, now we can;
`spibus = SPIBus(**{"cs": 9, "dc": 8, "sck": "GP10", "mosi": 11, "bl": Pin(13)})`
I started with 'not working' and now have so many options!
Would you prefer that this issue is left open until those commits are merged?

commit 3c1f1525 (2025-04-25): PimoroniBus: SPIBus fix unused pin arguments.

Fixes #1074.
commit eee6ee61 (2025-04-25): PimoroniBus: SPIBus fix unused pin arguments.

Fixes #1074.
commit 27d985c7 (2025-04-25): PimoroniBus: SPIBus fix unused pin arguments.

Fixes #1074.
commit a42a1221 (2025-04-25): PimoroniBus: SPIBus fix unused pin arguments.

Fixes #1074.
closed 2025-05-05 by jef41